adafruit/Adafruit_CircuitPython_HT16K33 #2: Get rid of "Get or set" methods

tannewt, 2017-12-07:
Get or set methods are a weird programming pattern inherited from MicroPython. It should be replaced with properties instead.

caternuson, 2018-05-16:
What's a good approach for using properties with 2D matrix like things? I think various syntaxes could be accommodated.
```python
pixels[x][y] = color
```
or
```python
pixels[x,y] = color
```

deshipu, 2018-05-16:
Both have their problems. In the first case, you have to create an intermediate object with its own getters and setters. In the second case, you have to create the tuple. That makes them slow and gives one more opportunity for memory fragmentation.

deshipu, 2018-05-16:
Also, it should be pixels[y][x], as it's arranged by rows first and columns second.

tannewt, 2018-05-16:
The Python `pillow` library does [x,y]: https://pillow.readthedocs.io/en/5.1.x/reference/PixelAccess.html

caternuson, 2018-05-19:
Works for me. Makes sense to follow suit with `pillow`. Any objections to going ahead and using this syntax for 2D property access?

deshipu, 2018-05-19:
I wonder if you would also want to have slices there for doing filled rectangles?

deshipu, 2018-05-24:
By the way, I would really recommend leaving the `pixel` method there, and only get rid of the blinking and brightness setters. There are many libraries that will work with anything that has a `pixel` method. Then again, you can probably just pass a framebuf object to them directly.

caternuson, 2018-06-18:
#11 did some of this for the `HT16K33` class and the matrix like displays. But still need to do something for the segmented displays.

Could do something like this, which would just be a simple refactoring of the current code:
```python
disp.text = "hello"
disp.number = 42
```
or maybe try and add something that acts more like `print()`:
```python
disp.print("hello")
disp.print(42)
```

deshipu, 2018-06-18:
I like the latter option much more.

tannewt, 2018-06-18:
+1 print

caternuson, 2018-06-21:
I think with #12 , this should be done. OK to close? #

tannewt, 2018-06-25:
Yup! Up to you.

closed 2018-06-25 by caternuson