adafruit/Adafruit_CircuitPython_HT16K33 #27: segments.py example missing display.show() or auto-write not working

jerryneedell, 2018-11-11:
A line with display.show() needs to be added at the end of segments.py to actually display the "12AB" message.

the display.print(42)  does auto-write but the dispaly of "12ab" does not appear unless display.show() is called.

Is this a bug with auto-write or an omission from the example?

auto-write should be enabled by default.

I found the same problem with display.scroll().
It did not auto-write. display.show() as needed.

jerryneedell, 2018-11-11:
strange -- looking at the history, it is not clear when the dispaly.show() at the end was dropped. The commit on Jul 18 added autowirite, but did not delete it. I can't find the commit that actually removed it.

caternuson, 2018-11-11:
I think this is an issue with forgetting to do a release after making some changes so it didn't get into the bundle. I should have done that. But it looks like adabot tagged a release a few days ago, so it should be in bundle now. Try with latest bundle release and see if that fixes it.

closed 2018-11-11 by jerryneedell