adafruit/Adafruit_CircuitPython_HT16K33 #29: Auto write causing flicker in some cases.

caternuson, 2019-01-28:
See here:
https://forums.adafruit.com/viewtopic.php?f=60&t=146815

Auto write is probably calling `show()` too often. Need to trace through and scrub code and make sure it only gets called as needed.

makermelissa, 2019-01-29:
I noticed one trouble spot is that it gets called in the `fill()` function, which is called in the middle of `_number()`. Perhaps save the current state of `auto_fill`, set it to false, do all the function calls for `_number()` call `show()` and then restore it to the saved state.

caternuson, 2019-01-29:
Good points. The auto-write was added later, so it was kind of shoe horned in. It's a bit of a mess and I only self assigned this issue since it's my mess. But if you want to PR something, please do.

makermelissa, 2019-01-29:
Sounds good, I'll go ahead and do that. I got really familiar with it when writing the latest addition to Adafruit_CircuitPython_FeatherWing.

caternuson, 2019-01-30:
Fixed with #30

closed 2019-01-30 by caternuson