adafruit/Adafruit_CircuitPython_HT16K33 #33: 7-segment Print display doesn't handle multiple periods

makermelissa, 2019-02-06:
I'm not sure if this is by design as the AlphaNumeric display handles this fine, but issuing a command like print with multiple dots such as in
```
import board
import busio
import adafruit_ht16k33.segments as segments

I2C_BUS = busio.I2C(board.SCL, board.SDA)
display = segments.Seg7x4(I2C_BUS)
display.print('0...9')
``` 
Results in it printing `0.9`

caternuson, 2019-02-08:
Not by design, looks like a bug. The second half of this conditional:
https://github.com/adafruit/Adafruit_CircuitPython_HT16K33/blob/master/adafruit_ht16k33/segments.py#L190
doesn't work for the 7 segment, so `scroll()` is not being called.
```python
Adafruit CircuitPython 3.1.2 on 2019-01-07; Adafruit ItsyBitsy M4 Express with samd51g19
>>> import board, busio
>>> import adafruit_ht16k33.segments as segments
>>> i2c = busio.I2C(board.SCL, board.SDA)
>>> disp = segments.Seg7x4(i2c)
>>> disp.fill(0)
>>> disp._push('.')
>>> disp.show()
>>> disp._get_buffer(7) & 0b01000000
0
>>> 
```
That same sequence for an instance of `Seg14x4` results in a non-zero return:
```python
Adafruit CircuitPython 3.1.2 on 2019-01-07; Adafruit ItsyBitsy M4 Express with samd51g19
>>> import board, busio
>>> import adafruit_ht16k33.segments as segments
>>> i2c = busio.I2C(board.SCL, board.SDA)
>>> disp = segments.Seg14x4(i2c)
>>> disp.fill(0)
>>> disp._push('.')
>>> disp.show()
>>> disp._get_buffer(7) & 0b01000000
64
>>> 
```

makermelissa, 2019-02-08:
Ok cool, I just wanted to make sure it was a bug before I worked on it.

makermelissa, 2019-02-08:
Yeah, you were correct. Thanks for pointing me in the right direction. The reason it wasn't is 2-fold. First buffer position 7 isn't used for this display. It's 8 for the 4th digit (see self._POSITIONS). Secondly the dot is lit with the highest bit (128). After that, a little experimentation and I got the fix working.

caternuson, 2019-02-08:
Done with #34 
thanks!

closed 2019-02-08 by caternuson