adafruit/Adafruit_CircuitPython_HT16K33 #36: Request for Seg14x8 class

pbricmont, 2019-04-14:
Could someone add a Seg14x8 class to segments.py? The chip supports 8 digits using the breakout or a custom board. 

Thanks,

Paul

ladyada, 2019-04-14:
@pbricmont do you have hardware to test against (we don't :)

pbricmont, 2019-04-14:
@ladyada I do: https://oshpark.com/shared_projects/XATdkjXm

I have three boards built. I got them working with Arduino by hacking the backpack library. The project is a multiline IOT display. 

[image]
[image]

ladyada, 2019-04-14:
@pbricmont how about you try making the variation and submit a PR :) you're the best equiped :D

pbricmont, 2019-04-14:
@ladyada I'll give it shot. I think I can see what needs to be done but I need to learn how to build it locally.

makermelissa, 2019-04-14:
The nice thing about CircuitPython libraries is you don't need to build them. They're just text like the regular CircuitPython scripts. I was under that same impression the first time I went to write one.

pbricmont, 2019-04-14:
@makermelissa So I can just replace the .mpy file with the .py in a library directory and it should work (assuming I've written some functional code)?

makermelissa, 2019-04-14:
Yes! That's exactly what I do.

makermelissa, 2019-04-14:
I should add the caveat that as long as your board has enough memory. I am partial to Feather M4 Express boards.

pbricmont, 2019-04-14:
@makermelissa Thank you! That makes it much easier. I've been testing with Metro M0 but I have Metro M4 and Grand Central available as well.

makermelissa, 2019-04-14:
Perfect. That should work great.

closed 2019-06-29 by pbricmont