adafruit/Adafruit_CircuitPython_HT16K33 #42: Provide method for setting via raw bitmask

caternuson, 2019-10-31:
Like the old library did:
https://github.com/adafruit/Adafruit_Python_LED_Backpack/blob/7356b4dd8b4bb162d60987878c2cb752fdd017d5/Adafruit_LED_Backpack/SevenSegment.py#L84

ref:
https://forums.adafruit.com/viewtopic.php?f=19&t=158151

ladyada, 2019-10-31:
@makermelissa will be giving this repo a scrubbing soon

caternuson, 2019-10-31:
Cool. Hopefully she can get further than my previous attempt :)
https://github.com/adafruit/Adafruit_CircuitPython_HT16K33/issues/8

makermelissa, 2019-12-30:
I think I got this figured out for the 7-segment display and 14-segment displays. Did we need this for other types of displays too?

makermelissa, 2019-12-31:
I have tested on the 7-segment and 14-segment featherwings as well as the large 7-segment display. Fixed via #43.

closed 2019-12-31 by makermelissa