adafruit/Adafruit_CircuitPython_HT16K33 #48: 16x8 Backpack bas different layout than Featherwing

makermelissa, 2020-01-15:
It appears the 16x8 matrix backpack is laid out differently than the 16x8. Running the exact same code (the demo code), I get 2 different results:
[image]
[image]I

We probably either need a new Matrix object (8x16?) or to pass a parameter in to differentiate how positions are calculated.

makermelissa, 2020-01-15:
Oops, photos updated.

caternuson, 2020-01-15:
I'd vote for two different classes.

makermelissa, 2020-01-15:
Thanks for the input, sounds good.

makermelissa, 2020-01-15:
Ok, I just wanted to see what was done in the Arduino library and there are 2 different classes Adafruit_8x16matrix and Adafruit_8x16minimatrix. Since we're using Matrix16x8 for the mini one and would prefer to not break code if possible, I'll probably use something like Matrix16x8Backpack or MatrixBackpack16x8.

closed 2020-01-15 by makermelissa

Docteh, 2021-08-13:
I ended up spending a bunch of time wondering about the why. As the schematics for both 16x8 boards have the modules identically arranged. I did find the answer, the location of the pins underneath the module:
Original module 
[image]

Newer Luckylight modules, posted July 2014
[image]

PR #49 (2020-01-15): Refactored Matrix classes. Added additional class for backpack
Fixes #48.