adafruit/Adafruit_CircuitPython_HT16K33 #57: Segmented Display Animations

makermelissa, 2020-01-28:
Not that set_digit_raw has been implemented, a fun little feature that might be to nice to add to this library   to take that a step further is animations. I'm imagining you passing a list of bitmasks and either a delay or list of delays to perform some animations.

geekguy-wy, 2020-02-06:
I can see where this could be useful, like for doing a status or progress bar or other things. How would you implement this by passing in a list of bitmasks? I can see how to do some stock things like progress bars by coding them but have never used bitmasks before. I am somewhat new to a lot of this stuff. I will research bitmasks.

makermelissa, 2020-02-07:
Take a look at https://learn.adafruit.com/adafruit-led-backpack/0-54-alphanumeric-circuitpython-and-python-usage for using set_digit_raw. now instead of passing one bitmask, you would pass a list of them along with a delay value and it would loop through the list. I hadn't thought out more details yet such  as whether it's for a single character or all characters. It was more of a place to put my ideas that I did have at the time.

geekguy-wy, 2020-02-07:
OK, I will look at that. I am currently thinking in terms of all characters of the display since you could just pass the character number if it is one digit you are working with. I will dig into this for sure because I have been working with one of these displays and it looks like an interesting and useful project.

geekguy-wy, 2020-02-07:
I am already getting some nice results! The function will have two to four parameters, with the last one or two being optional. The parameters will be (list of digit_numbers, list of bitmasks, delay between bitmasks, number of cycles). There is also the possibility that the second parameter can be a list of lists if the length of the list of digit_numbers is > 1. You may list each digit number more than once.

So, ([0, 1, 2, 3, 2, 1, 0], [0x2D3F], 0.2, 3) would apply the single bitmask to each digit in order with a delay of 0.2 seconds between bitmasks, and will cycle through 3 times.
There are some holes in this I still have to plug, but it will be cool when done!

This might not make sense right now, but I promise it will when you see code.

geekguy-wy, 2020-02-08:
Did you create the image that shows the 16-bit format of the segments?

geekguy-wy, 2020-02-13:
If PR #60 looks good and is merged, it should be time to look at rolling animate() into the library.

geekguy-wy, 2020-02-14:
#60 Fixes this issue

geekguy-wy, 2020-02-14:
PR#60 fixes this issue

makermelissa, 2020-03-07:
Already merged in. Closing.

closed 2020-03-07 by makermelissa