adafruit/Adafruit_CircuitPython_HT16K33 #62: Modify the brightness property.

geekguy-wy, 2020-02-12:
To make the library consistent with displayio, modify the brightness property so it uses a float instead of an int.

caternuson, 2020-02-12:
With a range 0.0 to 1.0:
https://circuitpython.readthedocs.io/en/latest/shared-bindings/displayio/Display.html#displayio.Display.brightness

geekguy-wy, 2020-02-12:
Got it, thanks. I already have the function to do the mapping from float to the range 0 to 15.

geekguy-wy, 2020-02-15:
@caternuson 
Does displayio carry the brightness property all the way through as a float? 
If so, do we want the ht16k33 library to do that or can we just carry it as a 0-15 int?

My thought is that if a user sets brightness as a float and we give them back an int that it might confuse things a bit.

makermelissa, 2020-03-10:
It's now merged in.

closed 2020-03-10 by makermelissa