adafruit/Adafruit_CircuitPython_HT16K33 #67: 8-digit seven segment display

lvajxi03, 2020-03-06:
Is there a tiny chance to support 8-digit 7-seg display? As we can see, HT16K33 pinout supports 8 digits, but...

makermelissa, 2020-03-10:
Hi, do you want to submit a PR? We don't have one to test with.

geekguy-wy, 2020-03-10:
I would love to do this, and it is one of my goals for the library. However, I do not have any of the 7-segment FeatherWings to test with.

geekguy-wy, 2020-03-27:
I would love to add support for multiple 7-segment, alphanumeric, and matrix displays. However, I only have one of the alphanumeric display Feather wings (Green) here. To do this and properly test everything, I would need at least three of each display. This is something I want very much to do. There are also the  Charlieplexed Feather wings and larger 16x9 Charlieplexed displays which are run by a different library I have not worked with yet.

makermelissa, 2020-03-27:
@geekguy-wy, I think he was referring to something like https://ebay.us/6qYcgl which uses a single ht16k33 chip as opposed to 2 4-digit displays at different addresses. I don't have one, which is why I requested he submit a PR.

geekguy-wy, 2020-03-27:
@makermelissa Yes, I know :) I was just expressing my desire to extend the support for Adafruit displays to allow the use of multiples in sort of a "chain." I thought that could be included in this issue, but I can open a separate one if you prefer that.

makermelissa, 2020-03-27:
I hear what you're saying and it's a great idea. I think a separate issue would be appropriate. :)

PR #101 (2022-06-21): Multidisplay support
Fixes #74. Fixes #67.

This has been tested on the FeatherWings. I have included examples. Library should function as before with single displays. The main difference is a lot of the hard-coded numbers are now calculated.
closed 2022-06-21 by dhalbert