adafruit/Adafruit_CircuitPython_HT16K33 #72: Pylint: arguments-out-of-order

dherrada, 2020-03-16:
The Pylint failure:
```
************* Module adafruit_ht16k33.matrix
adafruit_ht16k33/matrix.py:176:15: W1114: Positional arguments appear to be out of order (arguments-out-of-order)
```
The line in question
`176        return super()._pixel(y, x, color)`

Are these arguments actually out of order or should I add a disable?

Referencing main issue: adafruit/Adafruit_CircuitPython_Bundle#232
@kattni

tannewt, 2020-03-16:
This looks intentional to me and needs a pylint exception. @makermelissa would know for sure.

makermelissa, 2020-03-16:
Yep, disable please.

dherrada, 2020-03-17:
@makermelissa @tannewt Ok. I've disabled it. Thanks

kattni, 2020-03-17:
Closing this issue. PR is in process.

closed 2020-03-17 by kattni