adafruit/Adafruit_CircuitPython_HT16K33 #79: No LOL'ing with 7x4 Segment

Daviey, 2020-09-17:
Seg7x4 is feeling a bit unloved (:broken_heart: & :cry:) and could really do with a laugh to feel a bit better.  However, L,l,O,o are prohibited characters and silently dropped.

Please help put a smile on Seg7x4's face. :laughing:

tekktrik, 2021-11-22:
Is this still WIP?  I was thinking of implementing a way to add something like a character set to the display that could be used for a few reasons:

- To add characters not included in display hex numbers
- To pick how to display characters with multiple ways of showing them ("H" and "h" can both be valid in seven segment)
- To allow for displaying of characters that don't have clean seven segment displays (I've seen "w" be done in some convoluted way)
- To allow for non-English letters (Wikipedia shows some other alphabets)

I see a linked PR so I wanted to check in.

tekktrik, 2021-11-22:
Or would it be better to just open it up further under the current implementation, for letters like "L" and "O", but still prevent letters that don't have great representations, like "W"?

closed 2021-12-27 by FoamyGuy

tekktrik, 2021-12-27:
Thabka for the PR @FoamyGuy!  I may work on an implementation that allows for use of a character library as a solution for further letters, let m know if you have any thoughts.

FoamyGuy, 2021-12-28:
@tekktrik I like that idea, it would be nice if it had a list of letters that it is able to draw with the segments that it has and could use them with marquee.

tekktrik, 2021-12-28:
Sounds great!  I'll get on it in the new year!