adafruit/Adafruit_CircuitPython_HT16K33 #87: Error i2c with multiplexor

9acca9, 2021-03-13:
Hi.
Im using the ht16k33 after a multiplexor in the multiplexor is on the channel 2

So i always do this 
I2C_address = 0x71
bus.write_byte(I2C_address,2)

But, how i cant pass that to the:
"
example:
#matrix = matrix.Matrix8x8(i2c, address=0x70)
"

I try 
`matrix = matrix.Matrix8x8(I2C_address,2)`
That dont work i get:
```
File "/home/pi/.local/lib/python3.7/site-packages/adafruit_ht16k33/ht16k33.py", line 37, in __init__
    self.i2c_device = i2c_device.I2CDevice(i2c, address)
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_bus_device/i2c_device.py", line 50, in __init__
    self.__probe_for_device()
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_bus_device/i2c_device.py", line 151, in __probe_for_device
    while not self.i2c.try_lock():
AttributeError: 'int' object has no attribute 'try_loc
```k'

caternuson, 2021-03-13:
What is the multiplexer being used?

9acca9, 2021-03-13:
Sorry this was a dumb question. How can i delete this??
Thanks and sorry again

caternuson, 2021-03-13:
No worries. We can just close the issue. Seems like you've figure it out, but if you haven't, try posting the forums:
https://forums.adafruit.com/
That's a better place for general project help.

closed 2021-03-13 by caternuson