adafruit/Adafruit_CircuitPython_HT16K33 #91: Missing Type Annotations

FoamyGuy, 2021-09-27:
There are missing type annotations for some functions in this library.

The `typing` module does not exist on CircuitPython devices so the import needs to be wrapped in try/except to catch the error for missing import. There is an example of how that is done here:

```python
try:
    from typing import List, Tuple
except ImportError:
    pass
```

Once imported the typing annotations for the argument type(s), and return type(s) can be added to the function signature. Here is an example of a function that has had this done already:

```python
def wrap_text_to_pixels(
    string: str, max_width: int, font=None, indent0: str = "", indent1: str = ""
) -> List[str]:
```
If you are new to Git or Github we have a guide about contributing to our projects here: https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github

There is also a guide that covers our CI utilities and how to run them locally to ensure they will pass in Github Actions here: https://learn.adafruit.com/creating-and-sharing-a-circuitpython-library/check-your-code In particular the pages: `Sharing docs on ReadTheDocs` and `Check your code with pre-commit` contain the tools to install and commands to run locally to run the checks.

If you are attempting to resolve this issue and need help, you can post a comment on this issue and tag both @foamyguy and @kattni or reach out to us on Discord: https://adafru.it/discord in the `#circuitpython-dev` channel.

The following locations are reported by mypy to be missing type annotations:

- [ ] adafruit_ht16k33/ht16k33.py:36
- [ ] adafruit_ht16k33/ht16k33.py:48
- [ ] adafruit_ht16k33/ht16k33.py:59
- [ ] adafruit_ht16k33/ht16k33.py:72
- [ ] adafruit_ht16k33/ht16k33.py:89
- [ ] adafruit_ht16k33/ht16k33.py:102
- [ ] adafruit_ht16k33/ht16k33.py:110
- [ ] adafruit_ht16k33/ht16k33.py:125
- [ ] adafruit_ht16k33/ht16k33.py:128
- [ ] adafruit_ht16k33/segments.py:141
- [ ] adafruit_ht16k33/segments.py:152
- [ ] adafruit_ht16k33/segments.py:159
- [ ] adafruit_ht16k33/segments.py:164
- [ ] adafruit_ht16k33/segments.py:173
- [ ] adafruit_ht16k33/segments.py:188
- [ ] adafruit_ht16k33/segments.py:195
- [ ] adafruit_ht16k33/segments.py:200
- [ ] adafruit_ht16k33/segments.py:250
- [ ] adafruit_ht16k33/segments.py:273
- [ ] adafruit_ht16k33/segments.py:291
- [ ] adafruit_ht16k33/segments.py:309
- [ ] adafruit_ht16k33/segments.py:314
- [ ] adafruit_ht16k33/segments.py:325
- [ ] adafruit_ht16k33/segments.py:335
- [ ] adafruit_ht16k33/segments.py:363
- [ ] adafruit_ht16k33/segments.py:382
- [ ] adafruit_ht16k33/segments.py:390
- [ ] adafruit_ht16k33/segments.py:396
- [ ] adafruit_ht16k33/segments.py:413
- [ ] adafruit_ht16k33/segments.py:426
- [ ] adafruit_ht16k33/segments.py:435
- [ ] adafruit_ht16k33/segments.py:444
- [ ] adafruit_ht16k33/segments.py:455
- [ ] adafruit_ht16k33/segments.py:459
- [ ] adafruit_ht16k33/segments.py:470
- [ ] adafruit_ht16k33/matrix.py:23
- [ ] adafruit_ht16k33/matrix.py:32
- [ ] adafruit_ht16k33/matrix.py:36
- [ ] adafruit_ht16k33/matrix.py:41
- [ ] adafruit_ht16k33/matrix.py:83
- [ ] adafruit_ht16k33/matrix.py:91
- [ ] adafruit_ht16k33/matrix.py:99
- [ ] adafruit_ht16k33/matrix.py:107
- [ ] adafruit_ht16k33/matrix.py:115
- [ ] adafruit_ht16k33/matrix.py:153
- [ ] adafruit_ht16k33/matrix.py:168
- [ ] adafruit_ht16k33/matrix.py:185
- [ ] adafruit_ht16k33/matrix.py:198
- [ ] adafruit_ht16k33/matrix.py:208
- [ ] adafruit_ht16k33/bargraph.py:27
- [ ] adafruit_ht16k33/bargraph.py:34
- [ ] adafruit_ht16k33/bargraph.py:43

tekktrik, 2021-11-02:
Addressed by #92!

FoamyGuy, 2021-11-03:
closed by #92

closed 2021-11-03 by FoamyGuy