adafruit/Adafruit_CircuitPython_HT16K33 #98: colon @property on Seg7x4 hides multiple colon access on BigSeg7x4

angerer, 2022-02-09:
The @property that was added to the Seg7x4 class to mediate access to the self.colon = Colon() class member is interfering with using the member directly on the BigSeg7x4 subclass. The @property methods are being inherited from the superclass and only the self.colon[0] item can be accessed.

I'm happy to put together a pull request, but wasn't sure if you'd want to remove the @property methods from the superclass or override them in the subclass. If the property methods are going to be left, I would expect the self.colon member of the class to be renamed to self._colon to help clear up confusion about accessing it directly. (Though, I come from a Java background and I'm still learning the Python idoms and that might be incorrect.)

angerer, 2022-02-09:
I did some playing around trying to recreate this outside of my main project where I encountered the error. So far, I have learned:

- PyCharm doesn't see this as a problem when coding. I can't run it in the Mac environment, as I don't have the ability to execute the I2C objects required (pycharm interpreter at python 3.7.12)
- In a synthetic example with this type of inheritance it executes fine in PyCharm and on Raspberry Pi (superclass has self._var + @Property and subclass has self.var)
- A trivial program running on RPi that uses the BigSeg7x4 libraries fails. This is using the included 3.7.3 version of python, so maybe it's a python version issue. Details below:

```
import board
from adafruit_ht16k33.segments import BigSeg7x4

i2c = board.I2C()
bigdisplay = BigSeg7x4(i2c)
bigdisplay.colon[1] = True
```

results in:

```
Traceback (most recent call last):
  File "./7seg_property_test.py", line 10, in <module>
    bigdisplay.colon[1] = True
TypeError: 'bool' object does not support item assignment
```

dhalbert, 2022-02-09:
@makermelissa This is related to your #56.

The subclassing could be redone so that `class BigSeg7x4(Seg7x4)` is no longer the relationship, but instead there is an `AbstractSeg7x4` that contains the majority of the implementation, and then `colon` is specialized.

I am also thinking maybe it should be `Colons` and `.colons` for the general case, but that is an incompatible change. However, the `Seg7x4.colon` helper would still work, so that might be fine.

closed 2022-05-31 by makermelissa