adafruit/Adafruit_CircuitPython_HT16K33 #102: Shifting inverts red and green pixels on bi-color 8x8 matrix

MakerThornhill, 2022-08-04:
I'm using a 8x8 bi-color matrix with the HT16K33 Led Backpack on a QTPY RP2040 running CircuitPython 7.3.2 (2022-07-20).

I'm noticing red and green LEDs invert when shifting pixels by an odd amount. There's also an inconsistency between green and red when using the fill() method and when changing the colour of individual pixels. 

My code below highlights this behaviour (although it also occurs with the ht16k33_matrix_simpletest).

```
import board
import time
from adafruit_ht16k33.matrix import Matrix8x8x2

i2c = board.I2C()
matrix = Matrix8x8x2(i2c)

matrix.fill(1) # 1 should be red, but it sets all pixels green
matrix[0, 0] = 1 # Makes the 1st pixel red

time.sleep(1)
matrix.shift_up() # Shifts pixels up 1, colours are inverted so red = green, green = red
time.sleep(1)
matrix.shift_up() # Shifts pixels up 1, colours are inverted so red = green, green = red
time.sleep(1)
matrix.shift(0,2) # Shifts pixels up 2, colours remain the same
time.sleep(1)
matrix.shift(0,3) # Shifts pixels up 3, colours are inverted again so red = green, green = red

```

makermelissa, 2022-08-04:
Huh, interesting. That could be a bug that's been there a while that just needs to have the method overridden for that class.

BlitzCityDIY, 2023-02-16:
hello, I was testing with the new STEMMA version of the bicolor 8x8 matrix and noticed the identical behavior reported by @MakerThornhill above. I tested with a QT Py RP2040 and a Feather M4 on CP 8.0.2 stable. To make sure it wasn't my wiring, I tested with both boards with the Arduino library and the matrix behaved as expected.

makermelissa, 2023-02-21:
Fixed via #113

BlitzCityDIY, 2023-02-21:
hihi @makermelissa my PR actually only fixed the issue with color being flipped for calling matrix.fill(). The issue is still present with matrix.shift(). I wasn't able to make progress there

makermelissa, 2023-02-21:
Whoops, thanks for reopening.

BlitzCityDIY, 2024-03-13:
this is fixed with https://github.com/adafruit/Adafruit_CircuitPython_HT16K33/pull/118

closed 2024-03-13 by BlitzCityDIY