adafruit/Adafruit_CircuitPython_HT16K33 #104: Error setting individual Characters as numbers using Object as List

liltux, 2022-09-22:
Hello!  I was playing around with the 7 Segement x 4 backpack and QTESP32S2.  I was wanting to working running random numbers and have them stop on a random number sequence.  I was playing around with setting the individual characters as a random number between 0 and 9.  When trying to set as number I was getting this error:
[image]
But if I set the number as str(num) and set the character as a string, there is no error and it works.  Not sure if this is the desired operation or not.  I have not looked to close at the segments.py file as it is probably over my head and just wanted to share.

makermelissa, 2022-09-22:
Ok, it looks like the issue is that `def __setitem__(self, key: str, value: Union[str, int, float]) -> None:` can take an int or float in addition to a string, and then calls `def _put(self, char: str, index: int = 0) -> None:` which is expecting a string. I think `__setitem__` should explicitly be casting value to a string with `str()` and that should solve the error.

liltux, 2022-09-22:
OOPs, my brain was was not seeing the '' surrounding the values in the examples.  The desired usage is to use a string and not an integer.  That was my fault not reading clearly the information in front of me on the Adafruit Learn page [https://learn.adafruit.com/adafruit-led-backpack/0-dot-56-seven-segment-backpack-circuitpython-and-python-usage](url).  Thank you for looking at it anyways.  You know it could've been I was working on my Asus Chromebook 10" with beaglebone, text editor, web browser all open and toggling around and just couldn't see the small print.  :laughing:

makermelissa, 2022-09-22:
Either way, it's actually a bug, so I'd like to leave this open.

tekktrik, 2022-09-23:
Oof I think I really botched the typing on this repo...  It's just a typing fix right?

makermelissa, 2022-09-23:
> Oof I think I really botched the typing on this repo... It's just a typing fix right?

That would be the easiest fix, but allowing integers between 0-9 and converting them to strings would be another possible fix. Due to the size of the library, I think the typing fix would be preferable.

tekktrik, 2022-09-23:
Alright, I re-typed the library from scratch so hopefully it's better this time haha

closed 2022-09-23 by tekktrik