adafruit/Adafruit_CircuitPython_HT16K33 #108: 7x4 Characters Can be Added in for S2/S3

DJDevon3, 2022-10-22:
By default the 7x4 backpacks will only do 0-9, a-f, L, O. I realize this is probably by design for lower power boards.

[image]

I added characters so more of the alphabet will display on 7x4 segments (clock style) ... with a few missing logical characters like W, X, etc..  I've inserted dashes since ordinal wants at least something there.

Please don't penalize S2/S3 owners with removed characters just because lower end boards don't have enough RAM to use every character. Setup some kind of board specific detection switch?

The example sketch using base 16 or whatever is very hard to work with. Most people can do hex much easier. Recommend swapping out the bit strings for hex in customchar example at the very least. It's just too confusing to work with.

```py
NUMBERS = (
    0x3F,  # 0
    0x06,  # 1
    0x5B,  # 2
    0x4F,  # 3
    0x66,  # 4
    0x6D,  # 5
    0x7D,  # 6
    0x07,  # 7
    0x7F,  # 8
    0x6F,  # 9
    0x77,  # a
    0x7C,  # b
    0x39,  # C
    0x5E,  # d
    0x79,  # E
    0x71,  # F
    0x3D,  # G
    0x76,  # H
    0x30,  # I
    0x1E,  # J
    0x40,  # -
    0x38,  # L
    0x40,  # -
    0x54,  # n
    0x5C,  # o
    0x73,  # P
    0x67,  # q
    0x50,  # R
    0x6D,  # S
    0x78,  # t
    0x3E,  # U
    0x1C,  # v
    0x40,  # -
    0x40,  # -
    0x6E,  # y
    0x40,  # -
)
```
```py
if char in "abcdefghijklmnopqrstuvwxy":
            character = ord(char) - 97 + 10
```
Only changes required.

[image]

and now I have a usable alphabet running on my 7-segment clock displays on the esp32-s2.

makermelissa, 2022-10-27:
Hi, looks awesome. Want to submit a PR?

makermelissa, 2022-10-28:
Fixed by #109.

closed 2022-10-28 by makermelissa