adafruit/Adafruit_CircuitPython_HT16K33 #111: Incorrect handling of the "-" character on Seg7x4 displays

wkuranowski, 2023-01-30:
The problem is located on the following line:

https://github.com/adafruit/Adafruit_CircuitPython_HT16K33/blob/94453bf8fdcd0578a7957fc401ccc261ed46b68c/adafruit_ht16k33/segments.py#L464

Offset 16 for the "-" character is no longer valid. The closest valid offset is 20, but I don't know if this is a correct solution.

makermelissa, 2023-01-30:
You're right. It looks like this was caused by #109 which put a bunch of characters in front of the target character. It appears the "z" also took over the target character. I would add an additional character (so it won't accidentally have the formatting overwritten if somebody decides to make a z) and point it to that. So a new character would be 36 instead of 16.

makermelissa, 2023-01-30:
Want to submit a PR?

PR #112 (2023-01-31): Fix incorrect handling of the "-" character on 7x4 segments
Fixes #111
closed 2023-01-31 by makermelissa